BrewPi/firmware #8: Touch screen calibration is sometimes triggered, even when a calibration has been stored earlier

elcojacobs, 2015-03-17:
It should be investigated what causes this. Is the value in EEPROM invalid? Is it not stored correctly or not read correctly?

This is a problem when there is no physical access to the controller: while in the calibration screen, the controller cannot be controlled remotely. We could add a timeout, with a warning over serial.

m-mcgowan, 2015-04-07:
I investigated this with 2 external flash images from the community, and both were fine. 

- I erased external flash - powered on and the device wanted to calibrate. 
- Flashed each of the community images  to my device, the device did not need to calibrate, indicating the images were fine.

The suspected cause is a rogue 't' command being sent to the device, which will cause the device to jump into calibration mode. 

We will remove serial calibration mode in the next release when we have a boot menu to allow calibration.

m-mcgowan, 2015-04-10:
It turns out to be the script sending a 'T' command. For 0.2.8, we have removed this command since recalibration is available via tapping the screen during startup.

closed 2015-04-10 by m-mcgowan