BrewPi/firmware #13: User can interact with the embedded UI

m-mcgowan, 2015-04-10:
Scenario: the user can alter temperature setpoints and the mode using the embedded UI

Feature: user can initiate mode selection
Given brewpi running in non-test mode
When the user taps the mode region
Then a popup dialog of available modes appears.

Feature: mode selector
Given the user has initiated mode selection
When the user clicks on a mode, 
Then the mode is set
And when the user clicks outside the dialog,
Then the mode is not changed
 
Feature: user can initiate setpoint selection
Given brewpi running in non-test mode
When the user taps a temperature box
Then it changes to a temperature input box

Feature: user can change the setpoint 
Given a temperature input box is visible
When the up arrow is pressed, 
Then the temperature increases by the current temperature step, initially 0.1 degrees
And When the down arrow is pressed 
Then the temperature decreases by the current temperature step

Feature: user can increase the temperature step
Given a temperature input box is visible
When the left arrow is pressed, 
Then the temperature step is multiplied by 10, limited to a maximum of 10

Feature: user can decrease the temperature step
Given a temperature input box is visible
When the right arrow is pressed, 
Then the temperature step is divided by 10, limited to a minimum of 0.1

mirceageorgescu, 2015-12-08:
Any idea when these features are planned on the roadmap? Is there any way I can help out?

m-mcgowan, 2016-01-05:
We made a start on this in a feature branch `feature/touch-ui`, which you're welcome to take a look at. 

Our roadmap is focusing now on reworking the control module to support multiple PIDs, and the display will be rewritten to accommodate this, so we aren't planning on working on the existing display.