BrewPi/firmware #19: New system saves settings out the box

m-mcgowan, 2015-04-15:
Newly flashed devices will not store user settings without first having the eeprom initialized.

Currently, if the version stored in eeprom doesn't match the version expected by firmware, the eeprom is used in read-only mode - settings are not stored.  (We are missing notification to the user about this, but that's another issue.)

When the version is 0xFF (uninitialized) then the system will perform an eeprom reset to the current version. This ensures that settings will be stored from first use.

elcojacobs, 2015-04-15:
Perhaps change ==0xFF to > EEPROM_FORMAT_VERSION, in other words: any unknown version is considered invalid, not just 0xFF.

m-mcgowan, 2015-04-15:
if someone downgrades then they would lose their settings. 

The situation for having some other random value in there is if the arduino was used previously for something else. In that case I feel a manual eeprom reset is appropriate, and we add to the docs for first time programming.

closed 2017-06-22 by elcojacobs