BrewPi/firmware #54: fall back to beer sensor when fridge sensor is unavailable

elcojacobs, 2016-03-03:
The actuators in BrewPi are acting on fridge temperature. When this temperature is not available, it goes into IDLE.

We could create a new class, called FallBackSensor, which tries to read one sensor, and if it is unavailable, reads a second sensor.

This is a straightforward way to implement heating/cooling based on the beer sensor, when the fridge sensor is unavailable.

It will allow using beer profile mode with just one sensor and it will create extra redundancy for our normal use case with a fridge and a beer sensor available.

lucasjans, 2016-03-04:
I love this! Could you point to some existing lines of code so I could see how to implement?

elcojacobs, 2016-03-04:
I already got started: https://github.com/BrewPi/firmware/pull/55

I think there is not much work left, so this will probably be ready next week.