BrewPi/firmware #68: Making a change in the advanced settings sets "Cooler minimum ON time" 0

n8huntsman, 2016-08-14:
If I set my minimum on time, send to core, then update from core to verify setting, it appears correct.  However, making any other changes, such as max difference or Beer-to-Fridge proportional gain (Kp), then clicking update from core, resets the min on time to zero.

elcojacobs, 2016-08-22:
Bug confirmed.

Setting the beer-to-fridge max difference causes the minimum on time to become zero.

Tested on the Photon, 0.4.3

elcojacobs, 2016-08-22:
Related to what is fixed here:

https://github.com/BrewPi/firmware/commit/f96965e3ef065449a97883670555ad90b0e1ed37

The min cool time is in EEPROM/memory just after the max difference:
https://github.com/BrewPi/firmware/blob/master/app/controller/TempControl.h#L67

This is why it gets overwritten when applying the max difference with the wrong data length.

elcojacobs, 2016-11-23:
fixed in 0.4.4

closed 2016-11-23 by elcojacobs