BrewPi/firmware #86: Firmware 0.5.5 renders JSON with syntax error (unclosed quote).

NamelessCoder, 2018-05-20:
Hi guys,

Firmware 0.5.5 seems to render JSON with a syntax error which causes brewpi-www and others to be unable to read device lists or, when a list can be read, cause unexpected crashes in parsing of values.

The culprit:

`"Log2TempT`

```
 1 T:{"BeerTemp":5.56,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":28.88,"Log2TempT:{"BeerTemp":5.56,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":28.88,"Log2Temp,"State":4}
                                                                                                                                  ^
                                                                                                                                  Bad stuff happens here.
```

I'm going to attempt reverting to the firmware just before this one - but I think you should make this a high priority issue since automatic firmware upgrades appear to receive 0.5.5 when you perform them today (May 20th 2018).

elcojacobs, 2018-05-20:
That looks like a line was cut off prematurely, not just a missing quote.
And something I have not seen before. 0.5.5 didn't change anything related to JSON printing, so it is probably a problem with send or receive buffer size.
0.5.5 has been out for a while and is in use by many, so I expect the problem might be on the server side.

What device runs your BrewPi script?
How are they connected?
Are you running the latest version of the script?
Did you make any modifications?

I need more info, because this is probably due to USB buffers on Spark or server or perhaps even WiFi buffers/package length.

NamelessCoder, 2018-05-20:
Thanks for the very fast response @elcojacobs!

Here are the complete lines from stderr:

```
pi@raspberry:~ $ tail -f /home/brewpi/logs/stderr.txt
 May 20 2018 19:03:46   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 19:03:46   Line received was: T:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.94,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.94,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.94,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 19:04:26   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 19:04:26   Line received was: T:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2Temp4}
 May 20 2018 19:04:52   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 19:04:52   Line received was: T:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 19:04:57   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 19:04:57   Line received was: T:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 19:05:52   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 19:05:52   Line received was: T:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.88,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.38,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":29.94,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":5.31,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.81,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2Temp":null,"Log3Temp":null,"State":4}
```

I merely truncated the little example line since it appears to be the same problem repeating and always around the property name in question.

```
pi@raspberry:~ $ uname -a
Linux raspberry 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
```

Raspbian, not NOOBs edition (since I just use terminal anyway). Also reinstalled today, used to be the NOOBs edition.

All scripts have been updated around two hours ago which is also when I upgraded the firmware.

I had to manually change the installer to change `php5` to `php` and manually added `php-mbstring` which was missing. I doubt they're related though, since this appears to happen outside of PHP and the www interface runs just fine.

Connected currently via USB since I had problems with WiFi stability. I updated the firmware just today and the same brewpi installation that had worked began to exhibit this. It happens when I attempt to read the list of devices (with or without sensor values) and it does indeed also happen on `fermentrack` - which is what I was experimenting with earlier.

I did not do any special network tuning or other.

NamelessCoder, 2018-05-20:
I will switch to the NOOBs installation just to check if there are any differences perhaps in default networking options. I should have more info in around half an hour.

elcojacobs, 2018-05-20:
Raspbian should be fine.
I recommend using our docker images to install, as described on the wiki here:
https://wiki.brewpi.com/getting-started/raspberry-pi-docker-install

Should be much easier and quicker.

elcojacobs, 2018-05-20:
Another option is a crappy USB cable, try replacing it.

NamelessCoder, 2018-05-20:
Great tip about the USB cable - but unfortunately that doesn't seem to be it. Same cable now that I used for flashing it initially.

Reinstalled Raspbian Desktop from scratch, did nothing but set up keyboard then cloned the tools repository and did the install. Started it up, waited for the script to start and the same thing happens (when I view devices - it can collect temps for the probes I managed to assign earlier!).

Error log now contains, starting from blank file:

```
 May 20 2018 18:05:02   Notification: Script started for beer 'My First BrewPi Run'
 May 20 2018 18:05:02   Connecting to controller...
 May 20 2018 18:05:02   Background thread for serial started
 May 20 2018 18:05:02   Serial (re)connected for P1 [3e0052000851353532343835] at port: ttyACM0
 May 20 2018 18:05:04   Found BrewPi v0.5.5 build 0.5.5-0-g3a3680b9c, running on a Particle p1 with a V3 shield
 May 20 2018 18:05:04   Warning: version number of local copy of logMessages.h does not match log version number received from controller.controller version = 4, local copy version = 3
 May 20 2018 18:05:05   Cannot process line from controller: {"p":8,"m":[{"s":16384,"l":"m","vc":30,"vv":30,"f":"b","n":"0","v":201,"d":[]},{"s":262144,"l":"m","vc":30,"vv":30,"f":"s","n":"1","v":302,"d":[{"f":"s","n":"2","v":204,"_":""}]},{"s":262144,"l":"m","vc":30,"vv":30,"f":"s","n":"2","v":302,"d":[{"f":"s","n":"1","v":302,"_":""},{"f":"b","n":"0","v":201,"_":""}]},{"s":131072,"l":"m","vc":30,"vv":30,"u":"6EB73D53EC91DC1E722CC756A713936DBEC74EC324FA22BC8787E7227D73342A","f":"u","n":"1","v":5,"d":[{"f":"s","n":"2","v":302,"_":""}]},{"s":131072,"l":"f","vc":30,"vv":0,"d":[]}]}
 May 20 2018 18:05:06   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 18:05:06   Line received was: T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 18:05:19   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 18:05:19   Line received was: T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2Tempd:[{"i":0,"t":1,"c":1,"b":1,"f":9,"h":2,"d":0,"p":0,"a":"2882F2D5090000B1","j":  0.000},{"i":1,"t":1,"c":1,"b":0,"f":5,"h":2,"d":0,"p":0,"a":"28FE6BD509000010","j":  0.000},{"i":2,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,{"i":3,"t":1,"c":1,"b":0,"f":6,"h":2,"d":0,"p":0,"a":"2803F2FE08000023","j":  0.000}]
 May 20 2018 18:05:20   Available devices received: [{"a": "28799ED409000084", "c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 2, "j": 0.0, "p": 0, "t": 0}, {"a": "3A30C82A00000005", "c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 3, "n": 0, "p": 0, "t": 0, "x": 0}, {"a": "3A30C82A00000005", "c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 3, "n": 1, "p": 0, "t": 0, "x": 0}, {"c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 1, "p": 24, "t": 0, "x": 0}, {"c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 1, "p": 4, "t": 0, "x": 0}, {"c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 1, "p": 29, "t": 0, "x": 0}, {"c": 1, "b": 0, "d": 0, "f": 0, "i": -1, "h": 1, "p": 28, "t": 0, "x": 0}]
 May 20 2018 18:06:07   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 18:06:07   Line received was: T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempLog3T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.00,"Log2TempS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 18:06:34   JSON decode error: Expecting ':' delimiter: line 1 column 131 (char 130)
 May 20 2018 18:06:34   Line received was: T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2Tempd:[{"i":0,"t":1,"c":1,"b":1,"f": [... cut ...]
 May 20 2018 18:07:30   Line received was: T:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2TempT:{"BeerTemp":4.81,"BeerSet":2.00,"BeerAnn":null,"FridgeTemp":7.25,"FridgeSet":-4.00,"FridgeAnn":null,"Log1Temp":30.06,"Log2Temp":null,"Log3Temp":null,"State":4}
```

I have a 40A SSR and four temperature probes attached. Same setup used earlier.

Does the fact that it happens around character 130 every time give us any hints?

NamelessCoder, 2018-05-20:
Other views cause other errors - at other positions.

```
 May 20 2018 18:13:46   Line received was: C:{"tempFormat":"C","heater1_kp":30.00,"heater1_ti":1800,"heater1_td":60,"heaterS:{"mode":"b","beerSet":2.00,"fridgeSet":null}
 May 20 2018 18:13:46   Controller debug message: INFO MESSAGE 12: Received new setting: heater1_infilt = 1
 May 20 2018 18:13:47   Controller debug message: WARNING 1: Invalid command received by controller: {
 May 20 2018 18:13:47   Controller debug message: WARNING 1: Invalid command received by controller: "
 May 20 2018 18:13:47   Controller debug message: ERROR 10: Expected { got e
 May 20 2018 18:13:48   JSON decode error: Expecting ':' delimiter: line 1 column 84 (char 83)
 May 20 2018 18:13:48   Line received was: C:{"tempFormat":"C","heater1_kp":30.00,"heater1_ti":1800,"heater1_td":60,"heaterh:[{"i":0,"t":1,"c":1,"b":1,"f":9,"h":2,"d":0,"p":0,"a":"2882F2D5090000B1","j":  0.000},{"i":1,"t":1,"c":1,"b":0,"f":5,"h":2,"d":0,"p":0,"a":"28FE6BD509000010","j":  0.000},{"i":-1,"t":1,"c":1,"b":0,"f":5,"h":2,"d":0,"p":0,"a":"28799ED409000084","j":  0.000},{"i":3,"t":1,"c":1,"b":0,"f":6,"h":2,"d":0,"p":0,"a":"2803F2FE08000023","j":  0.000},{"i":-1,"t":1,"c":1,"b":0,"f":6,"h":3,"d":0,"p":0,"x":0,"a":"3A30C82A00000005","n":0},{"i":-1,"t":1,"c":1,"b":0,"f":6,"h":3,"d":0,"p":0,"x":0,"a":"3A30C82A00000005","n":1},{"i":2,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,"p":25,"x":0},{"i":-1,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,"p":24,"x":0},{"i":-1,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,"p":4,"x":0},{"i":-1,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,"p":29,"x":0},{"i":-1,"t":3,"c":1,"b":0,"f":3,"h":1,"d":0,"p":28,"x":0}]
 May 20 2018 18:13:48   Controller debug message: WARNING 1: Invalid command received by controller: k
 May 20 2018 18:13:48   Controller debug message:
 May 20 2018 18:13:48   Controller debug message: WARNING 1: Invalid command received by controller: }
```

Earlier I had a similar `Invalid command received by controller: 1` failure when I was trying to get `fermentrack` going.

NamelessCoder, 2018-05-20:
**Workaround found, issue still not solved or completely identified.**

Temporary update - I've worked around this issue by connecting over WiFi again, which seems to avoid the problem. Actually, WiFi is my preferred connection and I was only switching to USB due to the stability issues. So I'm hoping that the 0.5.5 firmware is stable on WiFi and I won't have a reason to struggle with the USB issues again.

You can give this issue low priority if you don't have other people experiencing it. And if you need me to provide more debugging information (or an SSH or VNC to a running system that does this live) just let me know! I also maintain open source projects so I'm quite happy to help ;)

Thanks for a nice and geeky product btw., I'm looking very much forward to maybe a stable WiFi so I can focus on geeking it up instead of making it run.

One last question, sorry for the Off-topic but I couldn't really find this in the docs:

Is there a way to interact with the controller, I assume with HTTP exchanging JSON, directly to the controller and if so is there a resource I could study to learn the API? I'd like to write small CLI tools to set temps, toggle actuators, read values and so on; nothing too complex.

elcojacobs, 2018-06-04:
It really does sound like a communication error on USB. All the lines that error are a combination of a truncated string concatenated with the next one.

It would be a good test to see if you get the same errors when you run BrewPi inside a docker container as described on the wiki. That's the new recommended install method, without the brewpi-tools repo.
You'll just have to deploy the container (with usb privileges in this case) and you're up and running.

Regarding the API: we're working hard on a next version that will have a REST API on everything. On the current version, it would be more of a hack and I recommend you don't waste your time on something that would be irrelevant in 2-3 months.